Xinyuan-LilyGO/T-QT #2: Board Manager URL  error

Vadimledyaev, 2022-07-11:
Hi, Arduino ide can not load your boards package :
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pagespackage_esp32_index.json

This url leads to **400: Invalid request** page

What is the correct url?
Thanks

LilyGO, 2022-07-13:
The URL in the description may have expired, please use a new URL

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

Vadimledyaev, 2022-07-14:
Thanks! The guide does not say which board should I choose?
There are plenty of boards available from the list **Tools - Boards - ESP32 -** . 
Which one is yours?

Sarah-C, 2022-07-24:
It'll be the : ESP32-S3 Dev Board                 

[Vadimledyaev](https://github.com/Vadimledyaev), be careful with the libraries!            
TFT_eSPI's current version does NOT compile with the Factory demo.         

For me, I don't use the library else where, so I just deleted it from my library folder, and copied https://github.com/Xinyuan-LilyGO/T-QT/tree/main/lib libraries into my lib folder.              

The alternative is a bit messier - copying the libraries from this github into the sketch Factory folder, flat. No subfolders. Then you can use the up to date versions in your other libraries.

I don't know what changed so much to break it - I didn't spend time looking at the errors. =/

JpEncausse, 2022-07-30:
Hi,
There is no ESP32-S3 Dev Board I choose Dev Module but got `Error compiling for board ESP32S3 Dev Module` I correctly copy /lib to Arduino /librairies folder
[image]

Sarah-C, 2022-07-31:
My mistake, as you saw, it's "Module"

@JpEncausse  
"Error compiling for board ESP32S3 Dev Module"..... there'll be more red/error text further up the text box. Can you dig it out? It'll detail file file name, line number, the line of code, and a little ^ arrow pointing to where it saw the issue.

JpEncausse, 2022-07-31:
Unfortunatly I didn't get more details, I assume it can't check for the given board,
Can you detail the Arduino option to set for T-QT ? (Memory, etc ...) to be sure I set evrything correctly (I tried many but still get the error)

LilyGO, 2022-08-02:
You just need to check "ESP32S3 Dev module" and the correct port. PSRAM needs to select OPI mode. Other configurations can be configured or set by default according to your project needs.

JpEncausse, 2022-08-02:
Ok so here is my configuration : 

**OS**
- Windows 11
- Arduino 1.8.19
- Copy T-QT-main/lib/* into C:\Users\{name}\Documents\Arduino\libraries

First error : The Factory example didn't run because the script look for lv_conf.h that must be in /librairies/lv_conf.h I copy it from elswhere. Now it is fixed

**Arduino** 
- Board : ESP32 Dev Module
- Upload Speed : 115200
- USB Mode : Hardware CDC and JTAG
- USB CDC On Boot : Disabled
- USB Firmware MSC On Boot : Disabled
- CPU Frequency : 240MHz (wifi)
- FlashMode : OPI 80MHz
- FlashSize : 8MB (64Mb)
- Partition Scheme : Default 4Mb with spiffs
- Core Debug Level : None
- PSRAM : OPI PSRAM
- Arduino Run on Core 1
- Events Run on Core 1
- Port COM4

Error 2 : So the `factory` and `lcd` script now compile but the board failed at reboot error with esptool (I assume the program is correctly write). On the board side, the screen stay black ...

LilyGO, 2022-08-05:
My mistake, I mistakenly thought this was a T-Display-S3. The PSRAM setting should be changed to QSPI or DISENABLE. (It depends on whether you bought the version with PSRAM or not). But selecting DISENABLE should be no problem.

LilyGO, 2022-08-05:
[image]

lynema, 2022-08-06:
After following this directions and compiling the factory example using the Arduino IDE and ESP32 library version 2.0.4, these lines are repeated over and over.

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> Saved PC:0x400454d5
> SPIWP:0xee
> mode:QIO, clock div:2
> load:0x3fce3808,len:0x43c
> ets_loader.c 78 

Board: ESP32S3 Dev Module 
Libraries were copied from the demo so the TFT library has the right config
USB A to C cable was used as well.  The program uploaded.

Are there other settings that are missing that need to be set?

LilyGO, 2022-08-06:
Please try to use ESP32 library version 2.0.4. And execute the simplest blink program to test the hardware function of the board. If the blink program runs normally, please use this routine: https://github.com/Xinyuan-LilyGO/T-QT/tree/main/example/lcd @lynema

lynema, 2022-08-06:
@LilyGO  
Tried this sketch with ESP32 library version 2.0.4.
```
void setup() {
	Serial.begin(115200);
}

void loop() {
	Serial.printf("Hello World");
	delay(3000);
}
```

Same thing happened.  The hardware is good.  When it was initially plugged in, the on-board display showed that it was attempting to connect to a wifi access point.

```
07:01:51.008 -> ESP-ROM:esp32s3-20210327
07:01:51.008 -> 
07:01:51.115 -> ESP-ROM:esp32s3-20210327
07:01:51.115 -> Build:Mar 27 2021
07:01:51.115 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
07:01:51.115 -> Saved PC:0x400454d5
07:01:51.148 -> SPIWP:0xee
07:01:51.148 -> mode:QIO, clock div:2
07:01:51.148 -> load:0x3fce3808,len:0x43c
07:01:51.148 -> ets_loader.c 78 
```

lynema, 2022-08-07:
It works under platform.io with a couple edits.

These lines need to be altered to account for case sensitivity in https://github.com/Xinyuan-LilyGO/T-QT/blob/main/example/factory/platformio.ini
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1
The board is esp32-s3-devkitc-1 which is different than what was mentioned above as the board name to try in the Arduino IDE. https://docs.platformio.org/en/latest/boards/espressif32/esp32-s3-devkitc-1.html

LilyGO, 2022-08-10:
@lynema Sorry, I rechecked my platformIO configuration. I am using arduino 2.0.3. version. Please recompile with this version.

LilyGO, 2022-08-10:
2.0.4 seems to have upgraded some features causing a reset, but I haven't checked its logs yet.

lynema, 2022-08-11:
@LilyGO Both 
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3
and 
platform_packages =  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.4
worked ok for me.

Vadimledyaev, 2022-08-11:
After a few days of trying to hook the device up to my PC(Windows10,  Arduini IDE 2) without success (probably a problem with drivers but even Zadig couldn't fix that) I managed to make it on my laptop. After that I tried a pretty fresh windows 10 virtual machine under my main one and that works, too.

So after all Arduino is working, confirmed. 

"SHA-256 comparison failed:" error is kinda normal:
```
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
```

P.s, in case you haven't worked with lvgl lib like I haven't, To make it work you must copypaste lv_conf.h provided by LiLyGO to a LIB folder, do not leave it as is in  "lib/lvgl/"!

Sarah-C, 2022-08-20:
>P.s, in case you haven't worked with lvgl lib like I haven't, To make it work you must copypaste lv_conf.h provided by LiLyGO to a LIB folder, do not leave it as is in "lib/lvgl/"!

Good to know! Thanks!

I've had the exact same issue with a couple of other boards on my desktop Vs laptop. ....... how curious....

ChrisJones79, 2022-11-17:
Using the USB-OTG board works for me but I had to zip the lib I wanted to use and added the file image.h to the sketch

lewisxhe, 2023-03-29:
Not active for a long time, will be closed

closed 2023-03-29 by lewisxhe

ChrisJones79, 2023-03-29:
Ok