Xinyuan-LilyGO/T-QT #5: Hour hand doesn't move as it should... fix included

Sarah-C, 2022-07-31:
If you're finding the hour hand in the factory demo to be a bit wonky... here's the fix.

The hour hand doesn't increment in fractions, so if it's 5 to 3, it looks almost like 5 to 2!

The fix is to also add on the minutes scaled to 1/12 of a circle... which is multiplying by 5.

lvgl_gui.cpp
```
void update_sensor(lv_timer_t *timer) {
  float volt = (analogRead(PIN_BAT_VOLT) * 2 * 3.3) / 4096;
  char buf[20];
  snprintf(buf, 20, "bat_volt:%.2f V", volt);
  lv_label_set_text(bat_label, buf);

  struct tm timeinfo;
  if (getLocalTime(&timeinfo)) {
    lv_img_set_angle(hour_img, ((timeinfo.tm_hour) * 300 + ((timeinfo.tm_min) * 5)) % 3600);  // The line that fixes the hour hand.
    lv_img_set_angle(min_img, (timeinfo.tm_min) * 60);
```

LilyGO, 2022-08-02:
thanks for your reminder.Please submit a pull request and I'll merge this.

Sarah-C, 2022-08-06:
https://github.com/Sarah-C/T-QT/commit/2980144b879a394d9c9399973fc774039b7895f6

[Micky513673326](https://github.com/Sarah-C/T-QT/commits?author=Micky513673326) has been nice enough to submit the pull request. Thank you!

Issue closed.

closed 2022-08-06 by Sarah-C