Xinyuan-LilyGO/T-QT #7: Waking up on right button - doesn't happen.

Sarah-C, 2022-08-06:
I see that the right button puts the device to sleep - which works fine.                     
However I note that esp_sleep_enable_ext0_wakeup is used to set the wake button to the same right button.            
This doesn't work either via a build in Arduino (latest regular build) nor in PlatformIO - the device stays off.

I've no idea where the error is, it doesn't look like it's how it's being used here - perhaps Espressifs own code?              
I'm just raising this issue for others to know it's a "known issue" - and not this codes fault.

main.cpp:
```

void go_to_sleep(void) {
  Serial.println("sleep");
  digitalWrite(PIN_LCD_BL, HIGH);
  gpio_hold_en((gpio_num_t)PIN_LCD_BL);
  digitalWrite(PIN_LCD_RES, LOW);
  esp_sleep_enable_ext0_wakeup((gpio_num_t)PIN_BTN_R, 0);
  esp_light_sleep_start();
}
```

Gargantuanman, 2022-10-14:
Im having the same issu thats because for some reason the PULL UP of the pin 0 fall when you send the controller to deepsleep that shouldnt be happening i believe there is some error on the schematic because supposely the button isn pulled up directly to regulator out, so it should be 3.3v always.

mmMicky, 2023-01-30:
As far as I know ESP32-S3 has this problem. You can use esp_sleep_enable_ext1_wakeup instead.

afpineda, 2023-02-09:
> Im having the same issu thats because for some reason the PULL UP of the pin 0 fall when you send the controller to deepsleep that shouldnt be happening i believe there is some error on the schematic because supposely the button isn pulled up directly to regulator out, so it should be 3.3v always.

Despite being externally pulled-up, maybe the GPIO is disabled before entering deep sleep mode. Check this code:

```c
void enterDeepSleep()
{
  // disable radios
  esp_bluedroid_disable();
  esp_bt_controller_disable();

  // Disable pins to avoid current drainage through pull resistors
  // Enable proper pull resistors for wake up (if available)
  for (int p = 0; p < GPIO_NUM_MAX; p++)
    if (GPIO_IS_VALID_GPIO((gpio_num_t)p) && !((p >= 6) && (p <= 11))) // Exclude flash memory pins
    {
      uint64_t pinBitmap = BITMAP(p);
      if (pinBitmap & ext1_wakeup_sources)
      {
        if (ext1_wakeup_mode == ESP_EXT1_WAKEUP_ANY_HIGH)
        {
          gpio_pullup_dis((gpio_num_t)p);
          gpio_pulldown_en((gpio_num_t)p);
        }
        else
        {
          gpio_pulldown_dis((gpio_num_t)p);
          gpio_pullup_en((gpio_num_t)p);
        }
      }
      else
      {
        gpio_pulldown_dis((gpio_num_t)p);
        gpio_pullup_dis((gpio_num_t)p);
      }
    }

  // enter deep sleep
  if (ext1_wakeup_sources != 0ULL)
  {
    ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON));
    ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(ext1_wakeup_sources, ext1_wakeup_mode));
  } // else reset is required for wake up
  esp_deep_sleep_start();

  // should not enter here
  log_e("power::powerOff(): Deep sleep not working");
  delay(5000);
  ESP.restart();
}
```

`ext1_wakeup_sources` and `ext1_wakeup_mode` are global variables. `ext1_wakeup_sources` is a bitmap of configured GPIO numbers to be used for wake up.

```c
// Power management
static uint64_t ext1_wakeup_sources = 0;
static esp_sleep_ext1_wakeup_mode_t ext1_wakeup_mode = ESP_EXT1_WAKEUP_ANY_HIGH;
```

[Source code](https://github.com/afpineda/OpenSourceSimWheelESP32/blob/main/src/common/power.cpp)

lewisxhe, 2023-03-29:
I discovered this issue during yesterday's reorganization and has now fixed it

closed 2023-03-29 by lewisxhe

Sarah-C, 2023-06-27:
lewisxhe - this was a PCB routing change?