Xinyuan-LilyGO/T-QT #10: Factory clock color

kirk5275, 2022-08-08:
How can I change the colors of the clock? Background, hands etc.?

kirk5275, 2022-08-09:
My Display is mostly gray with white letters. Second hand is yellowish orange.

LilyGO, 2022-08-09:
please attach your picture

kirk5275, 2022-08-09:
[image]

No matter what camera I use the is a blue tint to the picture. I see all gray with my eyes.

kirk5275, 2022-08-09:
[image]
Here is a better representation.

kirk5275, 2022-08-09:
[image]

LilyGO, 2022-08-10:
Abnormal display color is generally caused by several problems.
1. RGB and BGR display format settings are wrong.
2. The order of sending color buf is reversed.
3. The colors are set to invert.

  Select a normal color image to convert to an array and display the image using the provided TFT_eSPI library. If the display is normal, record the picture display format as RGB or BGR. reconstruct

kirk5275, 2022-08-10:
Just want to say that the duck looks okay. Green background white duck yellow beak and orange feet.

I am using your factory example with no modifications. Why should I have to change anything to get the colors to come out correctly? Also I don't know how to do do what you told me. 

Again might I have a bad unit?

LilyGO, 2022-08-13:
If duck color is green background white duck yellow beak and orange feet. Then its color is normal.

kirk5275, 2022-08-15:
Okay. I might have wasted your time and you mine. The one with blue clock is in \T-QT-main\lib\TFT_eSPI\examples\Smooth Graphics\Anti-aliased_Clock.

LilyGO, 2022-08-15:
This obviously has nothing to do with the hardware. It may just be that a register needs to be modified during initialization or that it needs to be high first when sending color BUF.

kirk5275, 2022-08-15:
I am closing this as I don't have a color problem

closed 2022-08-15 by kirk5275