Xinyuan-LilyGO/T-QT #8: Programming with PlatformIO

kirk5275, 2022-08-08:
In order to program I have to hold reset, hold IO0, release IO0 and release IO0. Then click on the upload icon in PlatformIO to get it to program.

I have a ESPRESSIF ESP32-S3-DevKitC-1 board and I don't have to touch anything. PlatformIO works using the same code

LilyGO, 2022-08-09:
T-QT is uploaded via USB. If you want to view serial information through USB, you need to enable USB CDC on Boot in the Arduino configuration page. If you need to add a flag in platformIO:
build_flags =
     -DARDUINO_USB_MODE=1
     -DARDUINO_USB_CDC_ON_BOOT=1

kirk5275, 2022-08-09:
I'm sorry but that was not my question. I am talking about uploading code. To Upload I have to do the following:
1. Press and hold reset
2. Press and hold IO0
3. Release reset
4. Release IO0
5. Click on upload icon in PlatformIO

I do not have to do those steps if I use a  ESP32-S3-DevKitC-1 dev board from ESPRESSIF 

I do see it executing a reset when it says connecting as my device does reboot. But it never goes into programming mode.

Uploading .pio\build\esp32-s3-devkitc-1\firmware.bin
esptool.py v3.3
Serial port com13
Connecting...  <-----------------------------------------------------------------**(RESETS HERE)**
Traceback (most recent call last):
Failed to get PID of a device on com13, using standard reset sequence.
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 5387, in <module>

    _main()
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 5380, in _main
    main()
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 4687, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,    
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 115, in get_default_connected_device  
    _esp.connect(before, connect_attempts)
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)        
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 629, in _connect_attempt
    self.sync()
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 519, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 458, in command
    self.write(pkt)
  File "C:\Users\krkjc\.platformio\packages\tool-esptoolpy\esptool.py", line 421, in write
    self._port.write(buf)
  File "C:\Users\krkjc\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
*** [upload] Error 1
========================================== [FAILED] Took 101.61 seconds ==========================================

 *  The terminal process "C:\Users\krkjc\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

LilyGO, 2022-08-10:
When you turn on the USB-CDC feature in settings. The next uploader can continue to use the port without pressing the button again. I have tested this function. The premise requires that you turn on the USB CDC on Boot function and do not use this pin in the program.

kirk5275, 2022-08-10:
I have these set in Platform.ini
build_flags = 
    ;-DARDUINO_USB_MODE=1
    ;-DARDUINO_USB_CDC_ON_BOOT=1

I am trying to upload the factory (clock) example. It does NOT work. I get the error in my previous post.
Using those settings does enable the Serial Monitor to work.

I am the one that is also having color issues. Could I have  bad hardware?

kirk5275, 2022-08-15:
Anything movement on this issue of not having to press buttons to upload in PlatformIO?

LilyGO, 2022-08-15:
I didn't need to press IO0 for every download like you did during PlatformIO development. I have stated that this flag needs to be added and do not do anything with the USB pins in the code.

build_flags =
-DARDUINO_USB_MODE=1
-DARDUINO_USB_CDC_ON_BOOT=1

If you think the board has a color problem. Please upload a TFT example and upload a picture for me to observe.

kirk5275, 2022-08-15:
I found I don't have a color problem the factory clock is only gray background.

I have set the build flags to what you told me. I am not doing anything with the USB pins. Like I said I can see the device doing a reset when it says connecting.

I have attached the simplest program to output on the serial port "Testing". It does not upload without me pressing buttons.

If you have a sketch you want me to try I will
[upload test.zip](https://github.com/Xinyuan-LilyGO/T-QT/files/9335727/upload.test.zip)
.

kirk5275, 2022-08-15:
I also tried the same code with Aurduino IDE with the following settings and it won't upload either

[image]

Sketch uses 233217 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13152 bytes (4%) of dynamic memory, leaving 314528 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3
Serial port COM13
Connecting...Traceback (most recent call last):
  File "esptool.py", line 5387, in <module>
  File "esptool.py", line 5380, in _main
  File "esptool.py", line 4687, in main
  File "esptool.py", line 115, in get_default_connected_device
  File "esptool.py", line 664, in connect
  File "esptool.py", line 629, in _connect_attempt
  File "esptool.py", line 519, in sync
  File "esptool.py", line 458, in command
  File "esptool.py", line 421, in write
  File "serial\serialwin32.py", line 325, in write
serial.serialutil.SerialTimeoutException: Write timeout
[26460] Failed to execute script 'esptool' due to unha
ndled exception!
Failed uploading: uploading error: exit status 1

Sarah-C, 2022-08-20:
The T-QT device works correctly when commanded to reset. I can confirm it works with PlatformIO:
```
Building .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v3.3
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM10
Forcing reset using 1200bps open/close on port COM10
Waiting for the new upload port...
Uploading .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v3.3
Serial port COM24
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:43:20:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x002d0000 to 0x002f8fff...
Flash will be erased from 0x00010000 to 0x0005ffff...
Flash params set to 0x023f
Compressed 15008 bytes to 10328...
Writing at 0x00000000... (100 %)
Wrote 15008 bytes (10328 compressed) at 0x00000000 in 0.3 seconds (effective 344.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 303.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 486.7 kbit/s)...
Hash of data verified.
Compressed 165488 bytes to 106746...
Writing at 0x002d0000... (14 %)
Writing at 0x002d863e... (28 %)
Writing at 0x002ddbbf... (42 %)
Writing at 0x002e4585... (57 %)
Writing at 0x002ea540... (71 %)
Writing at 0x002efe98... (85 %)
Writing at 0x002f529d... (100 %)
Wrote 165488 bytes (106746 compressed) at 0x002d0000 in 1.9 seconds (effective 700.1 kbit/s)...
Hash of data verified.
Compressed 326224 bytes to 169379...
Writing at 0x00010000... (9 %)
Writing at 0x00020d00... (18 %)
Writing at 0x00029f32... (27 %)
Writing at 0x00031e71... (36 %)
Writing at 0x0003786b... (45 %)
Writing at 0x0003cc00... (54 %)
Writing at 0x00041ec5... (63 %)
Writing at 0x00047780... (72 %)
Writing at 0x00050fb2... (81 %)
Writing at 0x00057f07... (90 %)
Writing at 0x0005d9a0... (100 %)
Wrote 326224 bytes (169379 compressed) at 0x00010000 in 3.3 seconds (effective 779.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

OP please check your device drivers, as the device correctly reboots into upload mode when commanded to by PlatformIO, without any user involvement.

kirk5275, 2022-08-20:
@Sarah-C Thanks for the suggestions. I tried it on a laptop computer and it worked. Play around with the USB drivers and got it working on the desktop computer.

closed 2022-08-20 by kirk5275

LilyGO, 2022-08-22:
https://github.com/Xinyuan-LilyGO/T-QT/issues/8#issuecomment-1215669544
As your Arduino is set up, turn on CDC ON BOOT

kirk5275, 2022-08-22:
I tried it on a laptop computer and it worked. Played around with the USB drivers and got it working on the desktop computer.
FYI you don't need these flags to upload. Only for the Serial Monitor to work. I just tried it.

 ;-DARDUINO_USB_MODE=1
 ;-DARDUINO_USB_CDC_ON_BOOT=1

Everything is working okay.

Sarah-C, 2022-08-22:
You're welcome! Happy coding!

evlo, 2023-01-26:
So could I connect RX TX to the usb data lines, keep 5V and GND and usb to work with uart like that?