Xinyuan-LilyGO/T-QT #14: Publish board settings for Arduino IDE

HowdyMoto, 2022-10-21:
Can you please publish what settings I should use here in the Arduino IDE?
[image]

LilyGO, 2022-10-26:
Just select ESP32-S3-DEV. Most of the other options vary based on individual usage.

mmMicky, 2022-11-23:
[image]

afpineda, 2023-02-09:
Note:

In order to **enter boot mode**, press and hold the `IO0` button while pressing the `RESET` button.

closed 2023-03-29 by lewisxhe

MichaelCastano, 2023-05-01:
> Note:
> 
> If required, in order to **enter boot mode**, press and hold the `IO0` button while pressing the `RESET` button.

Had problems connection on Windows 11 and that one hint solved it for me in the end 👍  Thanks
On MacOs it worked right out of the box following the Lilygo instructions.