Xinyuan-LilyGO/T-QT #16: What settings for factory example?

matterbury, 2022-11-23:
With the default settings compilation fails with:
```
Sketch uses 1379037 bytes (105%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41472 bytes (12%) of dynamic memory, leaving 286208 bytes for local variables. Maximum is 327680 bytes.
```
With Flash Size of 8MB (64 Mb) and Partition Scheme 8M with spiffs the device just crash reboots.

matterbury, 2022-11-23:
In fact it would be very helpful to have some idea of which examples work on the T-S3 board and what settings are needed for them. So far I'm about 3 for 8 working with the rest just blank screen or crash booting.

mmMicky, 2022-11-23:
T-QT is a flash size of 8M. T-QT PRO has 4M flash size and 2M PSRAM. You can set the partition table for storing programs slightly larger. E.g [image]

mmMicky, 2022-11-23:
[image]

matterbury, 2022-11-23:
Thanks, but as far as I can tell the factory example just does not work on the T-S3.
Many of the TFT_ ones do so I can play with those.

mmMicky, 2022-11-24:
It works fine on my computer. I wonder why your run fails. I need to fix it.

mmMicky, 2022-11-25:
I suggest you take a screenshot of the error or setting. I can help you check the problem.

mmMicky, 2022-11-25:
[image]
Here you need to set none

matterbury, 2022-11-25:
I set PSRAM to none and all the other settings as you have above.
Here is a video of the device while uploading, then a flash of the previous (clock) app when it rebooted, then the apparent crash looping after booting: https://www.dropbox.com/s/mhblk9lckomdqxq/IMG_0240.MOV?dl=0

Thanks for your help.

mmMicky, 2022-11-26:
I thought you had a problem with the T-QT product. If you have a problem with T-Display-S3, you need to enable the PSRAM-OPI function. It is recommended that you clone the repository without modifying it and use platformIO to compile. Because I used this platform to make. So I recommend you to use it.

mmMicky, 2022-11-26:
I can put the tested binary bin firmware here for you to burn and test.
[bin.zip](https://github.com/Xinyuan-LilyGO/T-QT/files/10095440/bin.zip)

matterbury, 2022-11-26:
Thank you very much, OPI PSRAM did the trick!

closed 2022-11-26 by matterbury