Xinyuan-LilyGO/T-QT #18: T-QT PRO circuit and charging IC

Vadimledyaev, 2023-01-29:
Hi, 

1. what charging IC is used in PRO version?
2. What is the charging current limit? I think it should be set to something like 20-30mA. Otherwise, a small battery will not last long.
3. Can you update the schematic pdf and provide the printed circuit of the PRO version?

Thanks.

mmMicky, 2023-01-30:
The charging IC used is TP4065. The charging current is limited to 400-500mA. I can update the schematic PRO version.

Vadimledyaev, 2023-01-30:
Well that current will definitely kill a small (60-200mAh +) battery and probably may cause **fire** and **explosion** . I strongly do not suggest to use it with a 1000mAh or less battery. (which size is huge).

For those who wanted a wearable device - desolder TP4065 and replace it with LTC4054 which charges the battery with currents from 10mA to 150mA (with Rprog set to 1000/Icharge, Ω, so from 100KΩ to 33KΩ is ok)

Weird and dangerous decision. Not supposed to be used in a wearable device. Liligo should kick in his butt the person who made the update without letting the people know the current.

mmMicky, 2023-01-31:
Adjust the charging current if necessary. No need to replace the chip. Only the current limiting resistor needs to be replaced. It's not always at 500mA charge when charging. When the battery voltage reaches about 4V, it will enter the trickle charging state. For specific correlation curves, please refer to the manual.

Vadimledyaev, 2023-01-31:
> Adjust the charging current if necessary. No need to replace the chip. Only the current limiting resistor needs to be replaced. It's not always at 500mA charge when charging. When the battery [...]

According datasheet, TP4065 Charging current is from 130mA to 1A. That means even if I replace the resistor, the charging current of 130mA is still too high for a small,  wearable device. It will destroy a 80mAh battery, sooner or later. But yeah, It might be ok for a big 200mAh ++ battery if you replace the resistor .

Vadimledyaev, 2023-03-01:
> The charging IC used is TP4065. The charging current is limited to 400-500mA. I can update the schematic PRO version.
HI, when would you update the schematic circuit?

lewisxhe, 2023-03-29:
Not active for a long time, will be closed

closed 2023-03-29 by lewisxhe

dskw, 2023-06-01:
> The charging IC used is TP4065. The charging current is limited to 400-500mA. I can update the schematic PRO version.

The schematic has not been updated, or anything else on the repo. It still does not even mention the QT-Pro and its charging capabilities.

@Vadimledyaev could you tell which resistor on the boards needs to be replaced and with which resistance or achieve 150mA charging current? I was not able to find an English manual for TP4065 yet ;(

lewisxhe, 2023-06-01:
@dskw  When I saw your message, I verified that there is indeed an error. I have changed the schematic diagram and related materials to T-QT Pro, and the different sizes of resistors I noted in the schematic diagram correspond to different charging currents. . PS: I am not the designer of T-QT, I have tried my best to make the schematic diagram easy to understand
https://github.com/Xinyuan-LilyGO/T-QT/blob/main/schematic/T-QT-Pro.pdf

dskw, 2023-06-01:
@lewisxhe ahh perfect, thank you :) You even included the resistor table <3

I've looked at both sides of my board and am even more confused now where that IC is supposed to be.
Looking under the LCD the label says it's a T_QT V1.3. Now I am trying to identify the RPROG location on the PCB.
I cannot find any IC on both sides that matches the 8-pin layout of the TP4065, only two ICs with 5 pins:
65b226 and 2U=D7C

Is v1.3 the "Pro" or are there more versions of the board around?
The Aliexpress page identifies the "Pro" based on the battery solder pad layout and going by that it should be a "Pro"

dskw, 2023-06-01:
You are a champ! Thank you so much, that was the information I was looking for!

dskw, 2023-06-01:
And even better, it is already at 10k per default, so the charging rate of V1.3 is at 130mA. No need to change anything for my 150mAh battery then

farwrc, 2023-12-19:
What is the battery specification for this board to charge direct from usb c?