Xinyuan-LilyGO/T-QT #19: Directly Wired Power 24V - Advice about wiring to 5V/3.3V pin

tyeth, 2023-02-04:
Can I just chuck 5V +/-5% into the 5V pin? I want to power the T-QT Pro from 24V, so was going to use an AP63205 which seems to fit the bill. I would appreciate any advice in this respect. I am aware the 3.3v model may be more appropriate, and could be put into the 3.3V pin probably safely, but I want 5V for other projects so will buy a few and therefore 5volts is more useful.

afpineda, 2023-02-09:
As far as I know, there are two answers: YES and...NO

**NO**

The built-in power regulator allows any input voltage from +5V to +7V more or less (don't know the exact value). The greater the voltage, the higher the wasted power and the higher the dissipated heat. In your case, to add a second power regulator is twice the power waste and much more heat.  **Not recommended**

**YES**

Going from +24V down to +3V3 means a lot of heat and wasted power. However, if you have no choice, just ignore the built-in power regulator at all, since the AP63205 is +3V3 output-capable. Wire such an output to `VDD3V` and `GND` as usual. Leave ' 5V' unconnected.

Be warned: **never attach `5V` nor the USB cable along with the external power source at the same time** or you will damage the devkit board.

**This is just my opinion. Didn't test myself**

lewisxhe, 2023-03-29:
I fully agree with @afpineda 's opinion,

closed 2023-03-29 by lewisxhe