Xinyuan-LilyGO/T-QT #25: Battery indicator and USB recognition for T-QT Pro

mrudick1, 2023-03-29:
Hello,

I would like to create a battery indicator that also recognizes if the T-QT Pro is plugged in and charging.  I've got the code sample from factory.ino which shows voltage from IO4, but that doesn't help recognize if the board is plugged in or not.  It also isn't a great indication of battery percentage left since it's nonlinear.

Does anyone have any example code on how to do this?

mrudick1, 2023-03-31:
Looking at the ESP32 doc, it looks like it's using GPIO19 for USB DM and GPIO20 for USB DP.  If plugged into my laptop I get an analogRead of GPIO19 that's <5 with no serial data.  When I use a charger/adapter to plug into and outlet, it looks like I get a steady reading of 4095 with or without a battery connected.  Now, I don't know if this is a suitable solution of inferring that a 4095 analog reading means the T-QT Pro is plugged in and presumably charging.

Any and all feedback is appreciated!

liangtongt, 2024-06-07:
where is the factory.ino? can you share it?