Xinyuan-LilyGO/T-QT #26: Which pins should used or I2C?

ismailagir, 2023-05-10:
Which pins should be used for I2C , is there any pre-defined ?

mrudick1, 2023-05-11:
It depends.  You can use the onboard JST connector as a Stemma/Qwiic connection.  If you choose to do so, the pins are:

GND = GND
3V = 3V
GPIO 44 = SCL
GPIO 43 = SDA

If you don't use this connector and solder it yourself, you can use any of the available GPIO pins for SCL and SDA.

jims1234, 2023-06-15:
Is there a 10K or equivalent pull up resistor mounted on the board on the SCL and SDA pins? if not do I need to add them externally?

lewisxhe, 2023-06-15:
It is not necessary to use a sensor module, which is generally integrated on the module. If a separate sensor is used, it is necessary

icubex, 2023-12-11:
It seems that on https://github.com/Xinyuan-LilyGO/T-QT/blob/main/image/pinmap_en.jpg the texts "IO44" and "IO43" should be exchanged.

ubidefeo, 2024-04-11:
@icubex 
I found out the same thing by just trying.

@icubex @mrudick1 I have opened #37 to address this

u.