Xinyuan-LilyGO/T-QT #27: Stuck with this board

Chanete, 2023-05-12:
Hi, I'm stucked with this board, any help would be appreciate. 

I've just bought 2 of thie tiny display from Ali, I've downloadad this repo and just tried to compile any of the samples to no avail. 

To focus in one I've tried to compila and download the Arduino_Life proyect. 

I've tried in Arduino Ide and in Platform IO with same results (nothing)

In Arduino IDE I just open the Arduino_Life.ino in samples dir. Select "ESP32S3-DevModule" board, select the proper com port and click upload. Everything goes great compiles, upload but, nothing happens. I've reset the device, power cyclo but nothing

If I try the same with platformio. there is a sligth difference, I've copied the esp-32-s3-qt-pro.json file to my boards dir. Then I compila and upload and sometimes it works, somtimes it stops around 40% of the upload. 

can someone give a clue or a a project (easy, just "Hello world..") that runs on this device.  I must be doing something real wrong, but I'm stuckedd...

Thx on advenac

mrudick1, 2023-05-15:
First, to put the QT into bootloader mode,  press and hold the left button while you press and release the reset button.

For working code, start here:  https://github.com/Xinyuan-LilyGO/T-QT

All the example programs work.

369Martin369, 2023-05-25:
PSRAM disabled?
8MB Flash?

mrudick1, 2023-05-25:
That would depend on which QT Pro you have.  There's one with just 8MB Flash and another with 4MB Flash and 2MB PSRAM.  Partition scheme can be default.  Also make sure you enable USB CDC On Boot.

jims1234, 2023-06-15:
I'm very glad you posted these suggestions above.. especially about " to put the QT into bootloader mode, press and hold the left button while you press and release the reset button." I just compiled and ran my 1st example "Rainbow" demo.. and tho there were several IDE warnings. related to the TFT_eSPI, it did load and run. Thanks again

farwrc, 2023-12-14:
I have the same here buy on ali too, i done everything what it on git and nothing show on display and de serial monitor is in looping but i flash the firmware and this works but sketch none copile upload and nothing

lewisxhe, 2023-12-14:
Have you selected the correct configuration according to the README?
https://github.com/Xinyuan-LilyGO/T-QT?tab=readme-ov-file#quick-start

farwrc, 2023-12-14:
> Have you selected the correct configuration according to the README? https://github.com/Xinyuan-LilyGO/T-QT?tab=readme-ov-file#quick-start

Yes, i did, i try all sketchs in this repository and nothing display, this is the serial monitor after flash the board with the firmware found in this link 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209ad42
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
E (233) psram: PSRAM ID read error: 0x00ffffff
Hello T-QT
Chip: ESP32-S3
ChipRevision: 0
Psram size: 0KB
Flash size: 8192KB
scan done

mrudick1, 2023-12-14:
Any time I've had this happen it's one of three things:

1) The board settings
2) Making sure you've chosen the correct board firmware (ESP32S3 Dev Module)
3) The User_Setup.h file in the the TFT_eSPI library has not been properly modified.

farwrc, 2023-12-14:
> Any time I've had this happen it's one of three things:
> 
> 1. The board settings
> 2. Making sure you've chosen the correct board firmware (ESP32S3 Dev Module)
> 3. The User_Setup.h file in the the TFT_eSPI library has not been properly modified.

What properly modification will need? copy the TFT_eSPI library from here

mrudick1, 2023-12-14:
When you update the TFT_eSPI library it overwrites what is there.  So your User_Setup.h file goes back to the default, which is not specific to the QT.  I suggest you update this library, then copy the /User_Setups/Setup211_LilyGo_T_QT_Pro_S3.h file to the root TFT_eSPI folder (one up) and then rename it to User_Setup.h.  Otherwise your code won't know what display you're using.

farwrc, 2023-12-14:
> When you update the TFT_eSPI library it overwrites what is there. So your User_Setup.h file goes back to the default, which is not specific to the QT. I suggest you update this library, then copy [...]

Ok i wil try that, but i dont update the library when the IDE ask i chose later

mrudick1, 2023-12-14:
It's a good idea to update the library, since bodmer has done a lot of work on it.  He also includes a QT Pro specific setup file for us..
I do advice that you first work using the arduino IDE because it's simpler to get started.  Once you have things working there, you can use platformio so you can localize the libraries and not have these kinds of problems moving forward.

farwrc, 2023-12-14:
All right i trying get the things work but none of the examples are work even a "hello world" apears on the screen, the board is functional because the firmware flashs and work and i can see the logo os lilygo and them the round clock only that still work but nothing tha ia upload work .....sorry my poor english i am from brasil

mrudick1, 2023-12-14:
Your english is great.  Don't worry about that.

OK, if you're seeing the Liligo logo on the screen, then it's not the TFT library.  Are you using the arduino IDE?  Have you chosen the correct port on your computer that your QT is attached to?  And have you allowed that port to access your computer (Mac will ask you to choose).
When you try to compile and upload, do you see the compile output complete and the upload begin to the QT?  It sounds like your QT only has the factory sketch on it.  So it might not be uploading at all.

farwrc, 2023-12-14:
> Your english is great. Don't worry about that.
>
> OK, if you're seeing the Liligo logo on the screen, then it's not the TFT library. Are you using the arduino IDE? Have you chosen the correct port on [...]

Yes i using a arduino IDE and chose the correct port i am use a PC wih win 10 the compile and upload run fine them when reset the board nothing show up

farwrc, 2023-12-19:
> Your english is great. Don't worry about that.
>
> OK, if you're seeing the Liligo logo on the screen, then it's not the TFT library. Are you using the arduino IDE? Have you chosen the correct port on [...]

Well, I'm no longer stuck with this board, on an Arduino forum some information was very useful and functional, resolving all the problems, not using the latest version of Espressif board management, version 3 doesn't work, the one that was solved for me was 2.011, and using your informations combined with this everything worked perfectly

mrudick1, 2023-12-19:
I'm so glad everything is working now!  Welcome to the QT family.  It's a remarkable little board.