Xinyuan-LilyGO/T-QT #28: LVGL_Factory crashes after color displays with error in LV_timer

369Martin369, 2023-05-25:
hi,

it runs 2 sec, then it crashes with error:
---------------------------
Hello T-QT
Chip: ESP32-S3
ChipRevision: 0
Psram size: 0KB
Flash size: 8192KB
[Error]	(9.610, +9610)	 lv_timer_create: Asserted at expression: new_timer != NULL (Out of memory) 	(in lv_timer.c line #172)
-----------------------------

any Idea???
Thx, M

mrudick1, 2023-05-25:
Give a little context.  Did you upload the factory sketch?  Or is this the first time you're turning it on?  Where does this error show, on screen or in serial monitor?

369Martin369, 2023-05-29:
yes, it is when I coplile factory sketch. output is serial monitor

lewisxhe, 2023-05-30:
See [QuickStart](https://github.com/Xinyuan-LilyGO/T-QT/blob/f298282b1837a90fad4f1722d082405742ea0077/README.md?plain=1#L25)