Xinyuan-LilyGO/T-QT #30: Blank Screen

robcloot, 2023-06-26:
I can upload the following sketch, and the serial monitor works, but nothing displays on the T-QT screen.

void setup()
{
 Serial.begin(115200); // Start Serial Monitor
 tft.init();           // Start tft screen
 tft.fillScreen(TFT_BLACK); // Black screen
 tft.setTextColor(TFT_WHITE, TFT_BLACK); // White on Black
 delay(10); // Pause for serial monitor to start
 tft.println("Hello World");
 Serial.println("Hello World");
}

void loop(){}

lewisxhe, 2023-06-27:
See [QuickStart](https://github.com/Xinyuan-LilyGO/T-QT/tree/f298282b1837a90fad4f1722d082405742ea0077#quick-start) and Tips

robcloot, 2023-06-28:
Yes lewisxhe, that fixed the problem.  I needed to  use the TFT_eSPI settings for T-QT in GitHub.  Thank you so much!

closed 2023-06-28 by robcloot