Xinyuan-LilyGO/T-QT #39: LVGL Factory example issue

JackIsGoofingOff, 2024-10-11:
Hi, I am here to seek for some helps about the lvgl factory example. I just ran the LVGL Factory example from the T-QT official folder and nothing wrong was reported from Serial Monitor. But the screen paused completely after a few seconds of running the wifi clock which I guess was related the following codes. May I know the possible reasons for it? 
`    
gui_init();
    while (1) {
      lv_timer_handler();
    }
`

lewisxhe, 2024-10-12:
The clock pause may be related to the wifi connection. Are you sure you are connected to the correct wifi?
https://github.com/Xinyuan-LilyGO/T-QT/blob/3f31e59a84d4530e26c236faab19391f80827593/examples/LVGL_Factory/lvgl_gui.cpp#L144

JackIsGoofingOff, 2024-10-12:
In the screen it shows that wifi has been connected and the clock actually can run for a few seconds

lewisxhe, 2024-10-14:
Try pressing the button to see if the page switches!

OomNiel, 2024-12-01:
Fix regarding this issue:

in 'void update_sensor(lv_timer_t *timer)' function:- comment out the line: 'if (!WiFi.isConnected())return;'
The WiFi will disconnect when time is retrieved from the NTP server.
This is set up in function: void timeavailable(struct timeval *t)
I presume this is done in order to prevent the application from spamming the NTP server with requests.

Also:
IF using the old T-QT board, add: '#define USE_HSPI_PORT' in the  'Setup211_LilyGo_T_QT_Pro_S3.h' file.
It seems that the board use the alternative HSPI spi port for driving the lcd.

The published example was obviously not tested for using on the old board.