Xinyuan-LilyGO/T-QT #40: USB host power

icubex, 2024-10-17:
I want to use the USB port in host mode. How can I set the USB connector to supply power when in USB host mode ? If not in software, how could I modify the board so that the USB port supplies power ? I am supplying power to the board via the 5V pin and don't need the battery feature.

lewisxhe, 2024-10-18:
It is not necessary to use battery power. The 5V pin interface and USB 5V are the same line. You can connect 5V from the 5V pin, so USBC will also have 5V output.

icubex, 2024-10-18:
I don't get any power from the USBC port after applying 5V to the 5V pin. The 3V3 pin provides 3V3 and the board is working correctly it seems because the loaded sketch is outputting data as expected. Are you sure the 5V pin is the same as the USBC 5V ? For reference I ran the same sketch on a ESP32S3 supermini board and its USBC port does provide power.

lewisxhe, 2024-10-18:
Oh! Sorry, the QT has a voltage switching function, so you can't use this function without removing the original part. If you need to transfer 5V to USB 5V, you need to short-circuit the diode D3.
[image]

icubex, 2024-10-18:
Excellent, it works. Thanks for the help !

closed 2024-10-21 by lewisxhe